s-leger/blender_cad_transforms #1: Numeric entries

renderthings, 2020-05-07:
Hi Stephen,
i test your addon using numeric entries as input values (as an architect I always use them), here are some problems i faced so far:

- G: all good

- R: clicking backspace reset the value except fo the first digit that still remains there (altough pressing a new value is accepted)

- S: same as R; it takes me some time to understand that while the value displayed is clearly a percentage, entering a numeric value is the "desired" value, so if i press on the keyboard 0.5 the selected element will not be scaled by 50 %, but by the percentage needed in order to have a 0.5 meter dimension..maybe having an option to choose the beahviour (scaling by percentage or scaling by desired value) would be useful

Here is my wishlist:

- G: enter x, y, z coordinate values, relative to the first click

- G, R, S: enter the minus symbol for "negativize" the value entered (as for the standard G shortcut)

Anyway thank you very much, i'm really appreciating it.

Ciro

s-leger, 2020-05-07:
Hi, 
Thank you for your feedback.
Negative values are allowed, but "-" must be specified at start. (even math operations are allowed like 2+3.5-9)
There is a bug with the display of values, not clearing the display.
ESC while entering values to keyboard clear the entire stack.
Agree for scale, 
by the way, scale still is a bit messy as blender doesn't allow not axis aligned scaling (unless you parent the object and use parent's scale). 

G allow keyboard entry when snapping to an edge, like place this at 1 meter from left corner of this edge. 
Relative coords for move are in the todo list.

renderthings, 2020-05-07:
Ok, thank you for your prompt reply!

s-leger, 2020-05-14:
Released 0.0.3

closed 2020-05-14 by s-leger