s-leger/blender_cad_transforms #2: Cannot enable the addon

counterpointer, 2020-05-07:
Error message (I'm on Mac, GPU not compatible)
"Traceback (most recent call last):
  File "/Applications/Blender.app/Contents/Resources/2.82/scripts/modules/addon_utils.py", line 351, in enable
    mod = __import__(module_name)
  File "/Users/username/Library/Application Support/Blender/2.82/scripts/addons/slcad_transform/__init__.py", line 49, in <module>
    from .slcad_transform import register, unregister
  File "/Users/username/Library/Application Support/Blender/2.82/scripts/addons/slcad_transform/slcad_transform.py", line 39, in <module>
    from .slcad_snap import (
  File "/Users/username/Library/Application Support/Blender/2.82/scripts/addons/slcad_transform/slcad_snap.py", line 29, in <module>
    from .snap_context import SnapContext
  File "/Users/username/Library/Application Support/Blender/2.82/scripts/addons/slcad_transform/snap_context/__init__.py", line 31, in <module>
    from .curve_drawing import GPU_Indices_Curve, gpu_Indices_enable_state, gpu_Indices_restore_state
  File "/Users/username/Library/Application Support/Blender/2.82/scripts/addons/slcad_transform/snap_context/curve_drawing.py", line 135, in <module>
    class GPU_Indices_Curve:
  File "/Users/username/Library/Application Support/Blender/2.82/scripts/addons/slcad_transform/snap_context/curve_drawing.py", line 137, in GPU_Indices_Curve
    shader = gpu.types.GPUShader(vert_3d, primitive_id_frag)
Exception: Shader Compile Error, see console for more details"

s-leger, 2020-05-07:
Hi, 
Thank you for reporting,
Could you report the error message found in the console ?

counterpointer, 2020-05-08:
After installation I cannot enable the addon, just getting this error message.

s-leger, 2020-05-08:
Would have expected some more information from console log as stated by "Exception: Shader Compile Error, see console for more details",  could you start blender from terminal using
`/Applications/Blender/blender.app/Contents/MacOs/blender
`
and try to see what's going wrong with the shader compilation.

counterpointer, 2020-05-08:
Sorry, I’m not an expert, just a Blender user.

Terminal report :

Registered Carver
----------------------------------------------
Archipack PRO 2.2.7 : ready 
----------------------------------------------
Warning: class MESH_OT_optiloops contains a property which should be an annotation!
/Users/username/Library/Application Support/Blender/2.82/scripts/addons/optiloops-master/__init__.py:318
    assign as a type annotation: MESH_OT_optiloops.angle_threshold
    assign as a type annotation: MESH_OT_optiloops.only_closed
    assign as a type annotation: MESH_OT_optiloops.keep_subsurf_influencing_loops
    assign as a type annotation: MESH_OT_optiloops.keep_seams
    assign as a type annotation: MESH_OT_optiloops.finish_dissolve
register_class(...):
Warning: 'EdgeWorksPanel' doesn't contain '_PT_' with prefix & suffix
Warning: class OBJECT_OT_ft_menu_options contains a property which should be an annotation!
/Users/username/Library/Application Support/Blender/2.82/scripts/addons/blenderbezierutils.py:8308
    assign as a type annotation: OBJECT_OT_ft_menu_options.optIdx
register_class(...):
Warning: 'TSCENE_PT3_layout' doesn't contain '_PT_' with prefix & suffix
register_class(...):
Warning: 'TSCENE_PT2_layout' doesn't contain '_PT_' with prefix & suffix
register
Warning: property 'release_confirm' not found in keymap item 'OperatorProperties'
ndof: unknown 3Dconnexion product c650

— After Optiloops & Edge roundifier disabled —

Registered Carver
----------------------------------------------
Archipack PRO 2.2.7 : ready 
----------------------------------------------
register_class(...):
Warning: 'TSCENE_PT3_layout' doesn't contain '_PT_' with prefix & suffix
register_class(...):
Warning: 'TSCENE_PT2_layout' doesn't contain '_PT_' with prefix & suffix
register
Warning: property 'release_confirm' not found in keymap item 'OperatorProperties'

s-leger, 2020-05-08:
Hmm nothing usefull about that shader complilation issue.
Will simply disable curve snap when things like that are going wrong, until i'm able to find the real source.

Oxervision, 2020-05-09:
I have the same error on Blender 2.83.15 & macos 10.13.6, this is the Console error:

```
GPUShader: compile error:
ERROR: 0:35: Use of undeclared identifier 'gl_FragColor'

Traceback (most recent call last):
  File "/Applications/Blender/BLENDER 2.83/Blender.app/Contents/Resources/2.83/scripts/modules/addon_utils.py", line 351, in enable
    mod = __import__(module_name)
  File "/Users/User/Library/Application Support/Blender/2.83/scripts/addons/slcad_transform/__init__.py", line 49, in <module>
    from .slcad_transform import register, unregister
  File "/Users/User/Library/Application Support/Blender/2.83/scripts/addons/slcad_transform/slcad_transform.py", line 39, in <module>
    from .slcad_snap import (
  File "/Users/User/Library/Application Support/Blender/2.83/scripts/addons/slcad_transform/slcad_snap.py", line 29, in <module>
    from .snap_context import SnapContext
  File "/Users/User/Library/Application Support/Blender/2.83/scripts/addons/slcad_transform/snap_context/__init__.py", line 31, in <module>
    from .curve_drawing import GPU_Indices_Curve, gpu_Indices_enable_state, gpu_Indices_restore_state
  File "/Users/User/Library/Application Support/Blender/2.83/scripts/addons/slcad_transform/snap_context/curve_drawing.py", line 135, in <module>
    class GPU_Indices_Curve:
  File "/Users/User/Library/Application Support/Blender/2.83/scripts/addons/slcad_transform/snap_context/curve_drawing.py", line 137, in GPU_Indices_Curve
    shader = gpu.types.GPUShader(vert_3d, primitive_id_frag)
Exception: Shader Compile Error, see console for more details
```

s-leger, 2020-05-09:
Thank you for reporting, things get a bit more clear now.
GPUShader: compile error:
ERROR: 0:35: Use of undeclared identifier 'gl_FragColor'

Oxervision, 2020-05-09:
When you fix the issue, please inform us.
Thanks in advance!!

PS: If you need that I test something on mac, tell it me.

s-leger, 2020-05-09:
if you are able to edit files, 
in /snap_context/curve_drawing.py replace the primitive_id_frag part by

`
primitive_id_frag = '''
uniform float offset;

in float primitive_id_var;
out vec4 fragColor;

void main()
{
    fragColor = vec4(offset + primitive_id_var, 0, 0, 0);
}
'''
`

Will close this issue once i do release an update so you'll notified.

counterpointer, 2020-05-09:
It works for me. Thanks.

Oxervision, 2020-05-10:
For me too, thanks!!

s-leger, 2020-05-14:
Released 0.0.3

closed 2020-05-14 by s-leger