s-leger/blender_cad_transforms #3: Version 0.0.3 frozen Blender interface

igiornotigi, 2020-05-18:
Hi, i installed the version 0.0.3 and after this when I select the tool and press G R or S Blender stop responding.

I don't have other origin/pivot plugin installed. If I remove the 0.0.3 and reinstall the v 0.0.1 version it works. 

Thanks

s-leger, 2020-05-18:
Any error in the console log ?
Could you share information about  os / blender version / graphic card ?

igiornotigi, 2020-05-18:
No error in console. If i still moving the mouse and click i can see the standard output about the operation, but all the "graphic"interface is frozen (viewport, menu, windows...). I can quit only with task manager.  
Blender 2.82a (same problem with other version)
gtx 1070

[image]

s-leger, 2020-05-19:
As far as i can see, there is an event stack overflow.
Modal body takes some time to complete - (~0.003 - 0.01 sec) and events fired in the between are stacked, so they run the modal again and again even after user interaction.

We are testing a version “consuming” events without actually evaluating the modal to keep the event stack as low as possible, to confirm this issue is fixed.

kobrad, 2020-05-21:
I have the same problem, the plugin randomly gets stuck when activated. There is no pattern to it, maybe it gets stuck more times than not. Nothing at all in the console, no way to predict when it will work and when it won't. It's exactly the same issue as igiornotigi reports.

igiornotigi, 2020-05-23:
Can it be related to the previous installation of the 0.0.1 version? Because in my notebook I installed the v 0.0.3 directly and it works well....

s-leger, 2020-05-23:
No, 
this issue is related to blender events rate, when the event firing rate is too fast for the modal to complete, events do stack and continue to run after interaction until the stack is empty.
It is fixed in latest release (0.0.6)
Will close this issue unless tests on 0.0.6 prove not to work for someone.
As side note, setup 0.0.6 and restart blender.

closed 2020-05-23 by s-leger