s-leger/blender_cad_transforms #4: When grabbing an object along an edge constraint, I cannot enter offset distance from reference end.

raindropsfromsky, 2020-06-06:
In your [demo video](https://www.youtube.com/watch?v=0ysfIw0ZojM&feature=youtu.be), I am not able to replicate what you do with **Cube 005** object in experiment starting at 3:45. In this experiment, you show that after snapping the movement to the edge, if we enter a number, the object is placed at that distance from the nearest vertex on the edge.

Well, I am not able to replicate this part: The object continues to show distance from its original position; and not the upper vertex of the edge. If I enter any distance, it seems to be from the original position, and not along the edge, counted from the reference vertex.

In fact, even your video also does not show that the Addon is measuring the distance from the upper vertex of the edge now. It always looks like the distance from the original poaition.

[image]

As shown in the screenshot, the on-screen display shows the distance along the red arrows (i.e., from the original position); not along the yellow arrow.

**Desired:** After the object snaps to the edge, it should start showing distance from the reference vertex. That way, the user knows that if he enters a particular value, it will be applied from the reference vertex, and measured along the edge.

s-leger, 2020-06-06:
Hi,
In order to move along an edge, move over the edge until the edge become red, then leave your mouse and enter the value on keyboard and press enter to confirm.
As soon as you move your mouse it does reset.
Display will always show the travel made from start point to end point as you may snap to an edge without actually enter any value, and if / when you enter the value by hand you'll the are certain of the distance.

raindropsfromsky, 2020-06-06:
Perhaps my description was not clear. Let me rephrase.
I have no problem snapping to the edge. Yes, it turns to red. Till that point, no problem.

Now the reference edge (to which the object is snapped) has two vertices.
According to your video, the vertex that is nearest to the object acts as reference.
I slide my object till my grabbing point (handle) snaps to that vertex.

At that point, I expect the screen readout to become zero. After that, as I slide my object along the edge, the readout should show distance along the edge from the reference point. In this case, it makes sense to enter a precise value instead of dragging the object along the edge.

However, the Addon does not behave this way:  The display keeps showing the distance from the original location of my handle. So it is not intuitive to enter a value that applies along the red edge, instead of along the calibrated black line that shows displacement from the origin.

s-leger, 2020-06-06:
Your issue is the "snap" to vertex basically disabling the edge mode.

When you snap over an edge, there is a small red circle over the nearest vertex and you may see the edge center (small crossing red line), so as soon as you are in one side of this red line you may enter value. But vertex and edge center do take precedence over edge snap when distance is under snap radius to allow to snap to vertex when enabled whatever other modes are also active.

Object's location will update while entering value.

raindropsfromsky, 2020-06-06:
I know that
(a) the edge looks (red dot at the end, and a cross at midpoint). 
(b) There is a  snapping distance. If the object moves away, it will stop being snapped to the edge.
(c) After I enter the value, the object will jump to the specified distance from the start point, along the edge.

I know all this, and I am happy with it.

I have made the following sketch to illustrate the issue I wanted to highlight:
[image]

s-leger, 2020-06-06:
May add the distance between origin and snapped point to edge widget so you have both.

raindropsfromsky, 2020-06-06:
In other words, the black calibrated scale should be temporarily suspended when the cub is snapped to the edge. During the snap mode, the OSD should start showing the cube's distance from the ref point. Thus, if I have anyways dragged the cube very close to the desired position, the OSD should be showing something like 4.998 m. 

At this point, I can release the mouse and enter the value 5 <ENTER>.

If I drag the cube away from the edge, the snap mode is discontinued, and I expect the normal black calibrated scale to show the distance from the original position once again.

raindropsfromsky, 2020-06-06:
> May add the distance between origin and snapped point to edge widget so you have both.

No, actually there is no need for a dual display. One display at a time is enough.
- When the object is being dragged freely, the black scale sows the distance.
- When the object is snapped to an edge, hide the scale and the distance from the origin. <br>Show only the dimension (along the edge) that can be entered from the keyboard.