s-leger/blender_cad_transforms #5: During grab operation, the normal flips to the wrong side, the user is unable to flip it back.

raindropsfromsky, 2020-06-06:
Not sure whether this is a bug, or I am doing something wrongly:

I am trying to land the "house" on a flat box. So initially I used pressed **E** (**Edge** mode). 
Then I selected the "house", and pressed G (grab) and released LMB. 
While dragging, I pressed **N** to enter **Normals** mode.

When the "house" nears the box, the normals flip every time. As a result, the "house" lands upside down. I cannot manipulate to keep the normal upright. I cannot flip the normal also.

The screenshot shows "before-after" picture. (The Addon's toolbar is also shown for clarity)
[image]

**Desired:**  
Let the user flip the normal manually.

s-leger, 2020-06-06:
This is more a WIP thing, and will be fixed in next release.

raindropsfromsky, 2020-06-06:
oh thanks!

closed 2020-06-06 by s-leger