s-leger/blender_cad_transforms #6: The numeric input field should be visible and editable

raindropsfromsky, 2020-06-06:
I repeated the Grab with Normals operation. 
My intent was to check if the numeric inputs can be entered and edited (if there is an entry error).

What I found is that this is not reliable: If I enter 2 digits, the "house" moves. But if I backspace an equal number of times, the "house" does not return to its original position.

The left screenshot shows the situation just after landing.
The right screenshot is after I entered a two-digit value, and then backspaced twice.

[image]

**Desired:**
Please provide an input box for the numeric value. 
That way the user knows what he entered. He can edit (correct) the value also.

s-leger, 2020-06-06:
hit esc / right mouse to cancel keyboard entry.
When you entry values, status bar display the entered values

raindropsfromsky, 2020-06-06:
Oh, didn't know that!

About the entry box- It would be nice to have an input box, something like the numerical values we enter in the main Blender GUI...

This is the input box for normal G command:
[image]

s-leger, 2020-06-06:
Transform depends on many contextual factors - snap target like edge / face / average / perpendicular, constraint origin axis / plane, and may even be stacked. So while it may work in some situations, it also may fails in many other.

raindropsfromsky, 2020-06-06:
> But think about it, numeric field is visible and editable, just not at same location.

Yes, that's my request: The feature is already there, but you could make it more stylish! :)
A low-hanging fruit...

closed 2020-06-08 by s-leger