s-leger/blender_cad_transforms #7: Doubt: Difference between "Normal" and "up"

raindropsfromsky, 2020-06-06:
In the demo video, at 4:37 you have moved a cube to a point on a plane, and then moved "up" by 1m.
But the plane has an inclination: It is not perfectly horizontal. So its Normal would be tilted.
So how come your command was able to pull "up" the cube by 1 m vertically?

Here, the expected normal is shown with a yellow arrow.
[image]

You entered a single digit (1). You did not use either {z and 1} or  {0, 0 and 1} to constrain Blender in z-direction and go 1 m.

Have I missed something?

s-leger, 2020-06-06:
Snapping to normal work like snapping to edge, when you enter a value you move from 1 in the direction of the normal, so distance from surface is 1

raindropsfromsky, 2020-06-06:
Yes, I know. But in which direction?

The plane appears to be inclined (not perfect horizontal). So its normal would also be inclined. Then your last command should go in the inclined direction (where the normal points).Instead, it goies in z direction. 

why is that?

s-leger, 2020-06-06:
Optical illusion, the normal red line is perpendicular to the surface (by definition). 
The plane is not straight some vertex are not aligned with x axis, but lie on xy plane.

raindropsfromsky, 2020-06-06:
Oops!!

closed 2020-06-06 by raindropsfromsky