s-leger/blender_cad_transforms #8: The use of "average" function (A key) is not clear in moving one point or a line segment.

raindropsfromsky, 2020-06-06:
The demo video contains two different manipulations, 
(a) From 4:42 to 5:30
(b) From 5:30 to 5:46

However, the underlying trick is not explained in both instances.

For example, in the first case, the Blender is in Edit mode, the CAD toolbar shows that two modes are active: Vertex selection and edge selection. But in actual grabbing of the point, neither seem to be working (there is no red circle around the point). What is going on here?  **[Issue#1]**

Then the middle point is grabbed (moved), and held above one end-point. Why? **[Issue#2]**

The **A** key is pressed. From description, I understand that this averages some points. But so far we selected only one point (point A). So, how is the second point added? (in the demo video, the A key is pressed only once.)

Finally, how does the A function find the point in the middle? And how do we take our junction to lie precisely there?

I can not see the blue, orange or pink circles at vertices.
*****************
In the second case, the three points are selected simultaneously, and taken to one end. Then key A is pressed. Then the selected set is dragged to the second end. It is not clear how this works, either.

A more elaborate example is needed for both use cases, with step-by-step explanations.

s-leger, 2020-06-06:
Take a look at this video, we still are polishing some rough edges, and will provide a complete tutorial along with 0.0.8 relase.
https://youtu.be/vhm_b-YVdK4

raindropsfromsky, 2020-06-06:
Y'know, while waiting for your response, I searched the net, and found this very video, and was actually watching it, when your reply arrived!

I also found [one more video](https://www.youtube.com/watch?v=TTKvxrgbFWU) by **Justin Geis**.

And in the process of searching, I discovered that you are the author of my most favorite Blender Addon: The inimitable Archipack Pro! 

And this addon is so beautiful that I am now your double fan!

Thank you indeed for making such wonderful products!

From your response and videos, I gather that this addon is still WIP, so I will postpone my feedback for later. Probably most of these teething issues would be taken care of.

Cheers!

closed 2020-06-08 by s-leger