s-leger/blender_cad_transforms #9: Fatal Crash - Requires Blender Restart for Plugin to work again.

arvindhal, 2020-06-08:
Hello,
I just discovered your add on when I saw it on a Youtube video and in the few minutes I was playing with it with some basic cubes I loved it! The potential seems endless. I was so excited I loaded a model I am currently working on and was using as the base for a game test scene. The model originally came from a Blender 2.7x file however I appended the model into a new 2.83 file and I've been working on it there.. The crash occurred in the 2.83 file. I am including the sanitized Blender screen shot of the crash with this report which follows. Thanks a million for this great plugin! I hope this report helps you make your product better. This tool is something Blender is sorely lacking!
-------------------------------------------------------------------------------------------
Crash Report
-------------------------------------------------------------------------------------------
Symptom: 
Fatal crash of the plugin that requires a Blender restart for the plugin to become functional again.

Cause:
Attempt to use the plugin by selecting any object in the scene or a new cube added to the scene.

Frequency:
Repeatable, happens every time.

Additional Factors:
I created two default cubes as can be seen in the image and attempted to use the plugin on those cubes thinking the crash may have been caused by Geometry issues. The crash still occurred with the default cubes.

OS: Windows 7
-------------------------------------------------------------------------------------------
[image]

s-leger, 2020-06-08:
Nurbs curves are not supported at this time.

arvindhal, 2020-06-08:
Thanks for the quick response! I understand. I've played around with it some more and you've done a fantastic job! I can't wait to get some more free time to put it through its paces in a production scene. If I notice anything else I will let you know. Thanks again for your efforts and excellent work.

closed 2020-06-08 by arvindhal