s-leger/blender_cad_transforms #11: Miscellaneous enhancements in s-CAD

raindropsfromsky, 2020-06-11:
[Note: I will describe the issue for **Grab**, but this may be applicable to **Rotate** and **Scale** also.]

**1. The distance shown with grab does not serve the CAD operations**
During the move, s-CAD displays the absolute distance. But in 3D space, showing absolute distance does not provide the precision required for CAD operations.  It should be defined in terms of [Cartesian coordinates](https://en.wikipedia.org/wiki/Cartesian_coordinate_system) (Δx,Δy,Δz) or in terms of [polar coordinate system](https://en.wikipedia.org/wiki/Polar_coordinate_system).

**2. When the dragged object interacts with another object, suppress the graduated scale**
When we start interacting with another object (inference/snap), we do not look at the distance from the original position. Therefore the scale becomes redundant at that time. It should be suppressed.

But suppose we stop interacting with another object, then the scale should re-appear.

To sum up, when there is a transition between "free move" and "inference/snap" modes, show it distinctly.

**3. The inference should be emphasized more.**
At present, when we snap to a vertex/edge, that target element is highlighted. But when we draw an inference, the inference line is not highlighted. 

For example, when we move along an edge, the drag mode still shows the black graduated scale attached to the origin and the distance from the original position. Instead, it should show a graduated scale that is anchored at one end of the scale, and distance from that end-point. (when the object is moved to the reference end, the distance should read 0). 

Another example: When we are locked to a Normal, then only a small line marks the normal. But once we start dragging there, s-CAD should show a stronger line (similar to the black graduated scale) from the base point of that normal. As we drag, distance from the surface should be displayed.

**4. The inference line should extend beyond the reference edge on both sides.**
When an inference is taken from an edge, it should extend infinitely on both sides. In other words, s-CAD should draw the inference line on both sides of the edge. (The distance on the other side of "zero" reference point is -ve.)

**5. In general, the display should show the same dimension(s) that c-CAD accepts as numerical value(s).**

This point is discussed in issue#4 at the end.

s-leger, 2020-06-11:
1 This feature exists in blender
[image]

2 You can't assume it is always what user want.

3 If you need precise feedback while grabbing using mouse (which is not precise by definition) do the move operation in 2 steps. 1 snap to corner, then move along edge.
As soon as keyboard entry is involved, you have a feedback through real-time update of location and precise move by entered values.

4 Inference line has direction, so when entering values on keyboard you know the direction. With infinite lines you loose this information.
In 0.0.8 when edge is involved in edge / constraint  intersection, parallel or perpendicular, you'll get feedback lines.

raindropsfromsky, 2020-06-11:
**1-3** Thanks for the tips and feedback!
**4** Thanks for the new "feedback lines" feature!

raindropsfromsky, 2020-06-11:
**3:** Unfortunately this 2-step (snapping+sliding along the edge) does not work for me.
Let me describe my results, for comparison:

I tried an experiment: I duplicated the default cube, and rotated its copy and scaled it down. 

In the experiment, I wanted to move the smaller cube with the 2-step move.
1. I could snap the smaller cube to the edge of larger cube
[image]

After snapping, the cubes look like this:
[image]

2. Then I slid the small cube along the edge of the main cube. This works only as long as I stay on the main cube's edge.
[image]

2a. But if I want to extend the inference line beyond the edge of the main cube, I lose the lock.
[image]

2b. Similarly, if I try to move the small cube on the other side, I lose the lock with the main cube's edge.
[image]

Am I missing something?