s-leger/blender_cad_transforms #12: Use the inference engine to draw lines and circles

raindropsfromsky, 2020-06-11:
With s-CAD, we suddenly realize that the 3D space becomes rich with a lot of reference points, edges and planes. 

Rather than just moving/rotating/scaling objects, can we also use the same references to draw lines and circles, like SketchUp? (If we can draw a circle, it means we can create any regular polygon by setting the number of sides).

Also allow the user to select two vertexes in 3D space and connect with them with a line (fill).

If this ability is added, we can draw complex 2D shape outlines on different planes, and then interconnect them to create a complex 3D shape. These "planes" need not be limited to x, y and z planes: We can use the curved surface of a cylinder, cone, torus, or icosphere.

s-leger, 2020-06-11:
This is not the purpose of CAD Transform add-on. 
Maybe for another add-on, but will not over complicate the code and usage - do less, but better.
CAD Transform is primary intended to fill a gap blender still has in the snap system until BF implementation is ready to land in production.
Once done, native blender tools probably will provide such ability out of the box.

closed 2020-06-11 by s-leger

raindropsfromsky, 2020-06-11:
> Once done, native blender tools probably will provide such ability out of the box.

Amen!

raindropsfromsky, 2020-06-11:
> This is not the purpose of CAD Transform add-on. Maybe for another add-on

This is exciting! I definitely look forward to a new 2D drawing addon!

With that, the users will find it very easy to draw 2D shapes in different planes. With some application of extrusion/Boolean, they will be able to construct very complex 3D shapes.