s-leger/blender_cad_transforms #15: When I press G to display grid during the grab operation, the status line should show [ESC] hides the grid

raindropsfromsky, 2020-06-11:
When I press G to display grid during the grab operation, the status line still shows that pressing G shows me the grid. Instead, it should show that pressing ESC hides the grid.

s-leger, 2020-06-11:
G enable / disable snap to grid.
ESC exit keyboard input / the operator

closed 2020-06-11 by s-leger

raindropsfromsky, 2020-06-11:
oops. I relied on Topic#2 in Jonathan's video (@ 5:10), which states that 

"while in grab mode, if we lock to a plane, we cannot tell which plane we are locked to. But if we press G, now we have a grid that shows where we are going." @ 5:20, it goes on to demonstrate this with SHFT+X plane. 

He does not mention any snapping to grid. And neither does his video look like it is snapping.

But more importantly, The status line shows G=Grid (not "snap to grid")
[image]

And when we press the G consecutively, it remains the same: It does not toggle between "snap to grid" and "free move".

Thus the instructions in the status lines do not match the functions of the G key.
Therefore, this is clearly a bug.

s-leger, 2020-06-11:
In status line it is impossible to depict in detail every single shortcut for obvious available space reason.
Detailed shortcut by category are available in the tool parameters.

G toggle the snap to grid mode, meaning when mouse is close to a grid intersection it will snap onto grid intersections, like any other snap modes. 

While snapping to grid, we draw the grid widget, but this grid widget may also be useful to see the working plane even when not snapping to grid, and this is what the trick he show is about.
 
I can't see any kind of bug there.

raindropsfromsky, 2020-06-11:
> In status line it is impossible to depict in detail every single shortcut for obvious available space reason.

The status bar is already much wider than the bottom of the screen. We have to scroll left+right to see the whole list. We can hover the mouse and roll the mousewheel to scroll left/right. Thus there is no limitation of space.

And here, the description just says "[G] Grid", which practically means nothing (compare that with "snap to grid" and "move free").
 
Just my 2 cents.