s-leger/blender_cad_transforms #16: Can't change the hotkey for rotation (and move and scale)

bzqp, 2020-06-17:
The "Keymap" panel in the "Add-ons" section doesn't have a hotkey setting for basic transformations (move, rotate, scale).

The global, Blender "Keymap" section finds these hotkeys for "CAD Transform" but doesn't let you to change them. It just ignores the new pressed hotkey and goes back to its default.

I would like to change my rotation hotkey to Q (carried from Sketchup), but can't unfortunately. The issue may be especially problematic for former Sketchup users, who as I understand, are one of the main demographics for the plugin.

Best Regards and thanks for the plugin! :)

s-leger, 2020-06-17:
This is a known issue with current tool setup in blender.
You must edit "slcad_transform.py" by hand
```python
class SLCAD_transform(WorkSpaceTool):
    bl_space_type = 'VIEW_3D'
    bl_context_mode = 'OBJECT'
    bl_idname = "slcad.transform"
    bl_label = "CAD Transform %s" % __version__
    bl_description = "Precise transforms operations\nShortcuts: G / R / S"
    bl_icon = os.path.join(os.path.dirname(__file__), "icons", "ops.transform.cad")
    bl_widget = None
    bl_keymap = (
        ("slcad.translate", {"type": 'G', "value": 'PRESS'}, None),
        ("slcad.rotate", {"type": 'R', "value": 'PRESS'}, None),
        ("slcad.scale", {"type": 'S', "value": 'PRESS'}, None),
    )
```
Change 'R' to whatever you want 
```python
("slcad.rotate", {"type": 'R', "value": 'PRESS'}, None),
```

bzqp, 2020-06-17:
Thanks, it works perfectly! :)

closed 2020-06-17 by bzqp