s-leger/blender_cad_transforms #25: Not snapping in Camera View

theoryshaw, 2020-08-30:
It seems to not be snapping consistently in camera views.  Is this a known problem, or am I missing something?

video: https://www.dropbox.com/s/zy4xd5yfd81wij3/2020-08-30_09-23-11.mp4?dl=0

Blender File:  https://www.dropbox.com/s/t6uer1placy9sbb/20200827-223_Randolph-render%20problem.blend?dl=0

s-leger, 2020-09-03:
There is probably an issue with value returned by blender api for view origin in ortho mode, however it does work in some situations, so realy hard to tell what's going wrong here and need some more investigations.

theoryshaw, 2020-10-26:
any luck tracking this down?  This happens all the time in camera view. Unfortunate, as I I'm in camera mode a lot via BlenderBIM's drawings.

s-leger, 2020-12-02:
Looks like an ortho camera location issue, where objects are not in front of camera (camera is at 0 altitude) ?
The addon use camera location and prevent snap to objects not in front, wich make sense when in perspective mode, but less in ortho mode.

theoryshaw, 2020-12-02:
Hi Stephan, 
I guess i'm not following you. 
[Here's](https://gitlab.com/openingdesign/223_randolph_st__atlanta/-/blob/e18365814fd6d35fc74481a10769dd6e96a4627a/Models%20and%20CAD/Blender/223_Randolph.blend) the file i'm working on and [here's](https://www.dropbox.com/s/9n27jzcb94jtvjx/2020-12-02_11-58-45.mp4?dl=0) a video capture of this scenerio. 
Cheers.

s-leger, 2020-12-02:
By default, blenderBIM create camera at 0 altitude, likely to be under the objects when view from top.
While objects are visible, they are not in front of camera so sanp can't work.
Moving camera over objects is likely to solve this issue.

theoryshaw, 2020-12-02:
Are you saying, move the camera cut so it intersections the objects, then it will work?

If so, this unfortunately, would be be pretty labor intensive ever time you want to move something. 

Perhaps i'm missing something. 

Or is this something BlenderBIM could fix?  

Pinging @moult

s-leger, 2020-12-02:
Move the camera so it is over all objects like you would do with a perpective camera

Moult, 2020-12-03:
I think something is up with the 3D viewport itself in the case of that file. Loading the file with "Load UI" turned off will fix it. Or switching to a different application template, or alternatively, merging the viewport with the timeline below (e.g. to delete the viewport), then splitting the viewport to create a new 3D viewport will also fix it.

The BlenderBIM Add-on drawings feature does set settings on individual viewports (e.g. for the ability to auto hide / show / set display settings for each drawing), and it has changed across BlenderBIM Add-on versions in a non-backwards compatible manner, so my current best guess is some of my code left it in a confused state of settings somewhere that broke the CAD transform somehow.

Not sure exactly what, though, but I don't think this is a bug with CAD transform, I think it is specific to this file, and most likely my fault :)

theoryshaw, 2020-12-03:
Thanks Dion for that screenshare, but I wasn't able to get that file to work on my end.  

I loaded without UI, as you did. 

I also appended into a newly created 'load factory settings' file.  

With both approaches, I was still having snapping issues in the camera views.

theoryshaw, 2020-12-03:
Phew, this one's hard to track down. 

I removed all blender programs and roaming data, and did a fresh install.  Everything is factory. 

I created [this](https://www.dropbox.com/s/iam10zk6jbttjy0/20201203%20-%20snapping%20in%20blender%20views.blend?dl=0) file, and still get snapping issues as illustrated in [this](https://www.dropbox.com/s/u59zh5wu7bfcyfi/2020-12-03_11-04-07.mp4?dl=0) video--some vertices show snaps, others do not.

Using Blender 2.91 w/ 0.91.0 Cad Transform

I did not install BlenderBIM at all.

s-leger, 2020-12-04:
Looks like blender has hard times to properly ray cast with ortho camera view and probably struggle with view origin. It inconsistently sometimes snap when i zoom on parts, and fails other times. 
The algorithm behind blender's view origin for ortho views rely on many tricky assumptions and may not be reliable when origin is not at center of screen, so ray_cast simply may fail.
Using a simple ortho view without camera still work as expected. So i guess in the meantime it is the way to go.

theoryshaw, 2020-12-04:
Thanks Steven, is this something that could be reported upstream to Blender developers?

If you report it, I can give it a +1. 

Cheers.

Moult, 2020-12-04:
@theoryshaw I can replicate the issue with your simple file. Yikes!

theoryshaw, 2021-04-17:
Hi Stephen, did you ever get a chance to report this upstream?  
I work in Blenderbim camera view all the time, and it would be nice to use your tool.

theoryshaw, 2021-05-22:
Hi Stephen, thanks for letting me test run 0.93.

For meshes, i've found it works pretty well!

For curves, however, it still seems buggy. 
Video: https://www.dropbox.com/s/1s17mgcz5fejszu/2021-05-22_06-58-06.mp4?dl=0

s-leger, 2021-05-23:
Fixed, -same wrong logick issue- will be in next release.

theoryshaw, 2021-05-29:
Hi Stephen, [here's](https://www.dropbox.com/s/j3adfzz96vomg9b/2021-05-29_09-24-23.mp4?dl=0) a video to help troubleshoot the problem.

closed 2021-06-05 by s-leger

theoryshaw, 2024-06-03:
Hi @s-leger, It seems that snapping while in camera view, is still an issue. 
Using 2.0.6
[Here's](https://hub.openingdesign.com/OpeningDesign/2917_International_Ln/src/commit/1973070d914933fec6122fac4ed74e82f6535499/Open/Models/BlenderBIM/temp.blend) the .blend file, if you need it. 

https://github.com/s-leger/blender_cad_transforms/assets/507113/583b77e2-b39a-4be9-977c-7761b20f0cd3

s-leger, 2024-06-03:
Would enable every available snap modes.

https://github.com/s-leger/blender_cad_transforms/assets/7442572/25e7db46-e05b-4d17-8770-45af5176acb8

s-leger, 2024-06-03:
Actually looks like a camera location / clip issue, with a camera located slightly further of the design it does work as expected. Also i use every available snap options.

https://github.com/s-leger/blender_cad_transforms/assets/7442572/6fe4e8cf-8264-438b-b73a-0622e6140c9d

theoryshaw, 2024-06-03:
> with a camera located slightly further of the design it does work as expected

What's the magic with that?  seems arbitrary.   I had to put the caerma at `z=50` for it to work.  What happens if there was something at  `z=40`, that obscured the lines at `z=0`?

s-leger, 2024-06-03:
My best guess is that blender struggle to find view origin wihen it is in orthographic mode, it does use some offset (based on clip) in true ortho views, but does not apply the same trick with cameras views. May also be related to large projection angle with very small focal distance leading to precision issue (?).

Would rather rely on 45° cameras for ortho mode so distance is half of desired view width.

theoryshaw, 2024-06-04:
Is this worth reporting upstream?