s-leger/blender_cad_transforms #27: snap still snapped to pre-modified geometry, as well, as the modified geometry.

theoryshaw, 2020-09-02:
Would be nice if the snap still snapped to pre-modified geometry, as well, as the modified geometry. 

video: https://www.dropbox.com/s/uitd0sadpas6ien/2020-09-02_11-53-22.mp4?dl=0

s-leger, 2020-09-03:
May add an option for this, but will have impact on performances at snap init time especially on huge mesh.

s-leger, 2020-09-19:
Looks like it is way harder than expected, snap use a ray cast to figure what's under the mouse pointer, but blender's raycast only see the final mesh result (with modifiers) and there is no obvious relation with non modified object's faces indexes - so no safe way to know the original - non modified - face hit by the ray.
The only solution i can think about is to use a pure opengl way to store non modified faces - as object's origin and curves, but it will be painfully slow on mouse hover (mostly a data structure memory initialisation performance issue) starting from medium mesh.

Michaelknubben, 2021-11-01:
Would it be an option to duplicate the mesh and display it as Wireframe only? Then after the transformation you'd delete that mesh.

s-leger, 2021-11-01:
The issue is the scene update timing occuring only once after operator's run.
So we add an object while operator run, try to snap, but blender's scene raycast is unable to see it as scene is not up to date.

Michaelknubben, 2021-11-01:
Hmm. Is this a limitation of the live tool? With operators within Pie Menu Editor for instance, I'd probably just run a duplicate command, then run the tool in question... is this not possible here?

s-leger, 2021-11-01:
you run 2 operators, with a scene update in the between. so 2nd one can see the dup object.

Michaelknubben, 2021-11-02:
Right, that's what I'm saying. Is this not possible for CAD transforms, for you to divide up your operator into multiple ones ran one after the other?