s-leger/blender_cad_transforms #28: default select method from 'tweak' to 'select box' while in 'Cad Transform' mode

theoryshaw, 2020-09-02:
Is it possible to change the default select method from 'tweak' to 'select box' while in 'Cad Transform' mode?  Or at least a preference option. 

Maybe I'm missing something.

[image]

s-leger, 2020-09-03:
CAD Transform is a "tool" so select tool is disabled when active.
Use "B" key to select using a box.

theoryshaw, 2020-09-03:
Thanks Stephen, 
Would it be possible to either have CAD Transform use the 'select box' as the default mode, or at least a preference one could set.  Just trying to avoid hitting 'B' for every action I do.

Michaelknubben, 2021-06-03:
It'd be ideal to have a dropdown to select what happens on drag, as in the default Blender Move, Rotate and Scale tools: 
[image]

s-leger, 2021-06-03:
Next version default to select box action without active transform operator.  
Will wait for users feedback / bug reports for 0.93.0 for like a month or so, and then relase 0.93.1

closed 2021-06-03 by s-leger

theoryshaw, 2021-06-27:
[image]