s-leger/blender_cad_transforms #30: Feature request: Please provide methods to scale by entering (a) desired magnification or (b) target dimension.

raindropsfromsky, 2020-09-22:
During scaling operation, the GUI offers two different figures: The % magnification and absolute dimension.

The user may want to enter the desired magnification (e.g. make this object double in size, means 200%).

Or he may want to say that resize the object till **this** particular dimension becomes x cm. 
The rest of the dimensions must be changed to maintain the aspect ratio.
In this case, he does not care what the original dimension is (or what the original object size is).

Please provide two different input methods to let the user transform the object.

s-leger, 2020-09-22:
Either check "absolute scale" option to input dimension, or add unit to the input value.
https://youtu.be/TWEoZghkalc?t=2272

closed 2020-09-22 by s-leger

raindropsfromsky, 2020-09-22:
Actually this does not address the second point I mentioned, where the user can lock the aspect ratio of the object while scaling; and he can enter a new value in one of the three dimensions. (Note that he will not enter a multiplier factor.)

The other two dimensions should automatically get multiplied by the same factor.

For example, assume that we have a cube with LxBxH dimensions of 2x3x5 units.
The user wants to make the Height equal to 7.5 unit.

Here, the addon calculates that the equivalent multiplier is 7.5/5=1.5.
So it multiplies all dimensions by 1.5.
Now the new dimensions of the cube is 3x4.5x7.5.

raindropsfromsky, 2020-09-22:
Since GitHub does not allow reopening of issues, I will raise a new bug. Thanks!