s-leger/blender_cad_transforms #35: Keep a ghost of the object you can snap/constrain to

theoryshaw, 2020-10-31:
Would be awesome UI feature if during a move/scale/rotate operation, there was a ghost of the original object that you could snap or constrain to.  And then after the operation is completed, the ghost obviously is deleted automatically.

[image]

theoryshaw, 2020-10-31:
Would, I think, help with...

- https://community.osarch.org/discussion/278/cad-transform-move-a-face-x-distance-from-a-point-onto-itself#latest
- https://community.osarch.org/discussion/277/cad-transform-constraining-along-an-edge#latest

s-leger, 2020-11-06:
May provide option to snap to self, without visible ghost. Actually there is code to avoid this behaviour as it make little sense (snap to selected elements when transforming in edit mode - snap to itself).