s-leger/blender_cad_transforms #37: Blender 2.91 API incompatibility?

Moult, 2020-11-30:
I didn't see anything about `ray_cast` here... https://docs.blender.org/api/current/change_log.html what did I miss?

I get this in 2.91 when I try to trigger the add-on.

```
Python: Traceback (most recent call last):
  File "/home/dion/.config/blender/2.91/scripts/addons/slcad_transform/slcad_transform.py", line 2392, in modal
    self._free_move(context, event)
  File "/home/dion/.config/blender/2.91/scripts/addons/slcad_transform/slcad_transform.py", line 2429, in _free_move
    slcadsnap.snap(context, event, grid_matrix)
  File "/home/dion/.config/blender/2.91/scripts/addons/slcad_transform/slcad_snap.py", line 797, in snap
    self._cast(context, hits, radius, use_center, deep_cast)
  File "/home/dion/.config/blender/2.91/scripts/addons/slcad_transform/slcad_snap.py", line 576, in _cast
    self._deep_cast(context, self._center, hits, deep_cast)
  File "/home/dion/.config/blender/2.91/scripts/addons/slcad_transform/slcad_snap.py", line 287, in _deep_cast
    direction
  File "/home/dion/.config/blender/2.91/scripts/addons/slcad_transform/slcad_snap.py", line 235, in _scene_ray_cast
    direction=direction)
TypeError: Scene.ray_cast(): required parameter "depsgraph" not specified

location: <unknown location>:-1
Error: Python: Traceback (most recent call last):
  File "/home/dion/.config/blender/2.91/scripts/addons/slcad_transform/slcad_transform.py", line 2392, in modal
    self._free_move(context, event)
  File "/home/dion/.config/blender/2.91/scripts/addons/slcad_transform/slcad_transform.py", line 2429, in _free_move
    slcadsnap.snap(context, event, grid_matrix)
  File "/home/dion/.config/blender/2.91/scripts/addons/slcad_transform/slcad_snap.py", line 797, in snap
    self._cast(context, hits, radius, use_center, deep_cast)
  File "/home/dion/.config/blender/2.91/scripts/addons/slcad_transform/slcad_snap.py", line 576, in _cast
    self._deep_cast(context, self._center, hits, deep_cast)
  File "/home/dion/.config/blender/2.91/scripts/addons/slcad_transform/slcad_snap.py", line 287, in _deep_cast
    direction
  File "/home/dion/.config/blender/2.91/scripts/addons/slcad_transform/slcad_snap.py", line 235, in _scene_ray_cast
    direction=direction)
TypeError: Scene.ray_cast(): required parameter "depsgraph" not specified

location: <unknown location>:-1
```

s-leger, 2020-11-30:
Download latest version (0.0.9) from github.
Blender's API bgl module was badly broken in 2.91 (there is a refactor to convert code into c++), and we where able to find some missing required functions - dumb like returning always 0 instead of true result.
As consequence snap to curve, lisolated vert/edge and origin are broken.

Moult, 2020-11-30:
Thanks :) Fixed in 0.0.9 beta!

closed 2020-11-30 by Moult