s-leger/blender_cad_transforms #41: crash when enabling addon

frenchgoodbear, 2020-12-16:
installed version 0.0.9 under blender 2.91 (ubuntu 20.10)
blender crashes when i click on enable tick in blender preferences window

stephenmm, 2020-12-20:
This was happening to me with version 2.8 ver of blender but when I updated to 2.91 it cleared up. I am also on windows so I am not sure that my comment helps you much/at all...

s-leger, 2020-12-23:
The 0.91.0 version does compile shader when you enable, crash at this point probably is related to a graphic card driver issue.
This issue is 2.91 specific and likely to be solved with next blender release.

frenchgoodbear, 2020-12-23:
So waiting for next release impatiently. I would be so glad to try and use your addon.
Thanks