s-leger/blender_cad_transforms #44: Loading CAD module n/w

bterstal, 2020-12-23:
When starting up Blender, the CAD module does not load (Icon not showing).  I need to go into Pref, disable to add-on and re-able it.  Using Blender version 2.91.0 and   CAD Transform 0.91.0

s-leger, 2020-12-23:
There is a known issue in some macs (depends on the hardware) as 0.91.0 fix a temporary regression in blender's 2.91 api. The regression will be fixed in next release of blender, but until then there is barely nothing i can do.

bterstal, 2020-12-23:
Thanks.  I appreciate the quick feedback.

Bernard

bterstal, 2021-01-05:
Just as a FYI..the "CAD Transform" is the only add-on that behaves that way.  The other add-on I have work fine.  Regardless, I use your add-on every day.

| Bernard

bterstal, 2021-01-10:
I also tried it on Blender 2.83.10.  Same issue.  Have to re-load every time I start Blender.  The configs are also lost (I always set the font to 35 and Widgets radius to 45), and I have to reset them. (I do manual "save Preferences").

s-leger, 2021-01-10:
Under 2.83 use version 0.8.0

bterstal, 2021-01-10:
Where can I get that version?  Do you have the URL for me to get it?

s-leger, 2021-01-10:
sorry, use 0.0.9 beta, downloadable from gumroad