s-leger/blender_cad_transforms #45: Auto-apply scale breaks instancing of objects

dburburan, 2020-12-25:
The plugin automatically applies scale by default. If the object has been instanced though, the mesh data is first duplicated before applying scale, breaking the instancing link.

[Feature Request]
Would it be possible to instead apply the scale transform to the mesh data directly, so that it propagates to all the instances?

I think this would improve most workflows and I can think of two reasons why it could be a good thing -

1. Blender by default won't allow this - if you try to apply scale to an instanced mesh, it will error with 'Cannot apply to multi-user object'. The user has to explicitly break the link first. Silently breaking instancing is therefore unintuitive and could catch users out.

2. Scale with auto-apply transforms as used within this plugin suggests a slightly different intent to Blender's normal scale tool. Quite often scaling is used here to set absolute dimensions - it's a kind of fast technique for placing vertices accurately, rather than wanting different scale versions of the same object.

If this were implemented and the user instead wanted the current behavior, they could first do Blender's 'Make Single User' command, followed by scaling, which would be on par with normal Blender workflow.

s-leger, 2020-12-25:
The scale operator is ment to allow scale along user defined axis (including hierarchy childs) - impossible with vanilla blender operator as scale appy along local object's axis.
In order to do this, it does align the object's transform matrix (and childs ones) to custom scale axis and to do so we must apply scale.

Now think about 2 "linked" objects, with parent-child relationship, but different rotations, when you try to scale the parent, the current behaviour allow to keep the overall shape by scaling both using the same custom axis, where keeping linked mesh will not.
I agree that doing so silently may be an issue, but i also feel that throwing errors on such simple operations where it is possible to handle things smoothly is somewhat more troublesome.

If you take a look at common issues newcomers are facing with, the "apply scale" comes near n°1 (as blender is the only one to do such things in the industry) and in facts the scale operation without apply is not well supported by blender's itself - and lead to modifiers like bevel inconsistency.

dburburan, 2020-12-28:
Ah I hadn't thought about the case with parenting, and I see now why it has to be this way.

Thank you for the thoughtful reply.

All the best,
David