s-leger/blender_cad_transforms #46: Is there a way to keep measurements in mm?

stillpointx, 2021-01-10:
Is there a way to keep measurements in mm?  It changes to cm after a certain distance

I'm using Blender 2.91 on Ubuntu 20.04 using CAD Transforms 0.0.9

See image below:
[image]

s-leger, 2021-02-14:
No, 
this behavior is intended to overcome precision issues in number rounding.

PsyonicOne, 2021-07-07:
I would also love this feature. Perhaps limiting it to 2 decimal places of the set unit?
I use your amazing addon to actually measure between snaps more than I do for moving, and even if the number displayed was rounded to 2 decimal places of a mm (or cm, or m, or ft, or what ever the user chooses as a base unit) this would be more than accurate enough to give me an idea of the gaps between objects that I don't want to actually move.