s-leger/blender_cad_transforms #48: Glitch with macOS 11.1 & Blender 2.93 alpha

nurb2kea, 2021-01-24:
Just to let you know, it works with 2.93.
But after a restart of blender the enabled add-on is enabled but not showing or working.

When clicking the addon to disable and then enable it will work agin until restart of blender.

thanks in advance 

sg

s-leger, 2021-01-24:
You should give version 0.8.0 a try.

nurb2kea, 2021-01-24:
there is only 0.0.9_beta and 0.91.0 available.
I use 0.91.0 and it has the same glitch with blender 2.90, 2.91, 2.92 on macos 11.1

doesn't matter where I install it. if it's default or userprefs.

s-leger, 2021-01-24:
Use 0.0.9_beta instead, 0.91.0 provide a hack for a blender's 2.91.0 specific issue compiling shader on the fly at startup and there are some issues with mac's hardware.

nurb2kea, 2021-01-24:
Bingo! 

Thanks a lot :-)

closed 2021-01-24 by nurb2kea