s-leger/blender_cad_transforms #53: Cad Transforms Not Loading Blender 2.92

bboldon, 2021-03-18:
Cad Transforms Not Loading Blender 2.92 on Mac.  After saving in Preferences with set-up changes when I closed and reopened Blender Cad Transforms does not load. I deactivate and reactivate and it works. I have to re-enter set-up changes each time I use Cad Transforms

Oxervision, 2021-03-21:
Yes, I have teh same issue. I obtain an error in Console when I run Blender 2.83 & 2.92 on macOS, this is the message:

Traceback (most recent call last):
 ```
 File "/Applications/Blender/BLENDER 2.83/Blender.app/Contents/Resources/2.83/scripts/modules/addon_utils.py", line 351, in enable
    mod = __import__(module_name)
  File "/Users/Oxer/Library/Application Support/Blender/SCRIPTS/addons/slcad_transform/__init__.py", line 54, in <module>
    from .slcad_transform import register, unregister
  File "/Users/Oxer/Library/Application Support/Blender/SCRIPTS/addons/slcad_transform/slcad_transform.py", line 49, in <module>
    from .slcad_snap import (
  File "/Users/Oxer/Library/Application Support/Blender/SCRIPTS/addons/slcad_transform/slcad_snap.py", line 33, in <module>
    from .snap_context import SnapContext
  File "/Users/Oxer/Library/Application Support/Blender/SCRIPTS/addons/slcad_transform/snap_context/__init__.py", line 31, in <module>
    from .drawing import GPU_Indices, gpu_Indices_enable_state, gpu_Indices_restore_state
  File "/Users/Oxer/Library/Application Support/Blender/SCRIPTS/addons/slcad_transform/snap_context/drawing.py", line 187, in <module>
    class GPU_Indices:
  File "/Users/Oxer/Library/Application Support/Blender/SCRIPTS/addons/slcad_transform/snap_context/drawing.py", line 192, in GPU_Indices
    primitive_id_frag,
  File "/Users/Oxer/Library/Application Support/Blender/SCRIPTS/addons/slcad_transform/snap_context/utils_shader.py", line 81, in __init__
    check_shaderError(self.program, bgl.GL_VALIDATE_STATUS, True, "Error: Program is invalid:")
  File "/Users/Oxer/Library/Application Support/Blender/SCRIPTS/addons/slcad_transform/snap_context/utils_shader.py", line 46, in check_shaderError
    raise Exception(errorMessage)
The addon icon doesn't appear in the Toolbar, the addon preferences doesn't show, If I deactive the addon and then I activate it, the icon appears anf the preferences too.
```
I hope you can help me with this issue.

Regards and thanks in advance!!

Addon version 0.91.0
macOS 10.13.6 - Blender 2.83 & 2.92

s-leger, 2021-03-21:
Download and setup 0.0.9 version instead.

Oxervision, 2021-03-21:
I'm going to probe with 0.0.9 and I'll inform.

Oxervision, 2021-03-21:
> Download and setup 0.0.9 version instead.

Thanks, with 0.0.9 version the addon works fine.

bboldon, 2021-03-21:
Thank you for the help, downloaded and loading fine!