s-leger/blender_cad_transforms #54: Scaling causing small rotation transformations

SaitoPhoenix, 2021-03-22:
Blender Ver. 2.90.1
CAD Transform Ver. 0.0.9
Default startup blender settings
Default CAD Transform add-on settings

When scaling a simple cube, snapping from one corner to another, the scale operation is causing a slight rotation.  This can occur even if the scale is not applied.  In most cases it is small (i.e. 0.0009 degrees), but it is compounding and seems to grow over multiple transformations.  I've had some objects that were off by more than a degree.  In some cases, the moment I click on the 2nd point for the scale, the object rotates before I've applied or cancelled the action.

[image]
[image]
[image]

s-leger, 2021-03-24:
This is a price to pay for custom axis scale in blender, as object axis is rotated on the fly to match custom scale axis. Precision issues occurs in the between, so you see that 0.00009° difference (wich is ~10 cm at 10km). Unfortunately such rounding issues are unavoidable in single precision context.

KarlZeilhofer, 2021-06-05:
I see this issue also as a noticeable problem. 
I've put some cubes together, with very basic scaling and rotation of 45°. 

The wall segments don't fit together any more precisely. 
[image]

Note: the bottom vertex is snapped together, but the top vertices have a distance of about 5mm for a wall with 2.78m height. 
[image]

I think, there is also some skew on the cubes. 
I don't know how sketchup is dealing with this rounding problems related to single precision float, but I've never experienced such deviations there. 

Is there any strategy in the usage, so that the problem gets minimized? (e.g. Going from large to small or better from small to large?

KarlZeilhofer, 2021-06-05:
PS: a scale operation never applies a rotation to an object, right? If so, then the object's angles could be corrected after the scale is applied.

s-leger, 2021-06-05:
In order to provide scale operations on custom axis (not aligned with object's space) it is mandatory to pre rotate the object's matrix so it does fit with the custom axis, then apply the scale transformation and revert rotation.
This also apply to hierarchy of parent / child objects.

Precision issues occurs, but are like  1e-6°  ~10cm at 10km / 0.01mm for 1m, barely unnoticeable at the scale of your model, so i guess your rotation issue is elsewhere, probably the constraint axis was not properly snapped.