s-leger/blender_cad_transforms #58: Active tool menu not shown

bitfabrikken, 2021-05-21:
Hi, I can't get the menu with the various icons to show. 
Hasn't worked in the last several blender versions I've had installed.

I've tried both slcad_transform_0.0.9_beta and slcad_transform_0.91.0 from gumroad.

OS: Linux 5.10.0-6-amd64 - Debian GNU/Linux 11 (bullseye) 
X: Xfce 4.16 on X.Org 1.20.11
Blender: 2.93.0 Alpha

Screenshot of what my viewport looks like with cad transform selected - no menu in top left corner, and very few things in the active tool context menu on the right:
[image]

Also get an error if I use G while having a vertex selected:
```python
Python: Traceback (most recent call last):
  File "/home/dan/bin/blender/2.93/scripts/startup/bl_ui/space_view3d.py", line 51, in draw
    self.draw_tool_settings(context)
  File "/home/dan/bin/blender/2.93/scripts/startup/bl_ui/space_view3d.py", line 68, in draw_tool_settings
    tool = ToolSelectPanelHelper.draw_active_tool_header(
  File "/home/dan/bin/blender/2.93/scripts/startup/bl_ui/space_toolsystem_common.py", line 782, in draw_active_tool_header
    draw_settings(context, layout, tool)
  File "/home/dan/.config/blender/2.93/scripts/addons/slcad_transform/slcad_transform.py", line 1042, in draw_settings
    km = context.window_manager.keyconfigs['blender addon'].keymaps['3D View Tool: Cad Transform'].keymap_items
KeyError: 'bpy_prop_collection[key]: key "blender addon" not found'

location: <unknown location>:-1
```

Skaifer, 2021-05-21:
Similar behavior here. 

Blender 2.93.0 Beta, Windows 10
Portable version installed with "Blender Launcher"
Tried both versions of addon

No menu items and tool options. Movement is working though. But when I switch addon off and on it shows this error:

 `Python: Traceback (most recent call last):
  File "C:\Portable\Blender\Builds\daily\blender-2.93.0-beta+blender-v293-release.a294670bacb1-windows.amd64-release\2.93\scripts\startup\bl_ui\space_view3d.py", line 51, in draw
    self.draw_tool_settings(context)
  File "C:\Portable\Blender\Builds\daily\blender-2.93.0-beta+blender-v293-release.a294670bacb1-windows.amd64-release\2.93\scripts\startup\bl_ui\space_view3d.py", line 68, in draw_tool_settings
    tool = ToolSelectPanelHelper.draw_active_tool_header(
  File "C:\Portable\Blender\Builds\daily\blender-2.93.0-beta+blender-v293-release.a294670bacb1-windows.amd64-release\2.93\scripts\startup\bl_ui\space_toolsystem_common.py", line 782, in draw_active_tool_header
    draw_settings(context, layout, tool)
  File "C:\Users\7xrav\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\slcad_transform\slcad_transform.py", line 1042, in draw_settings
    prefs = context.preferences.addons[__package__].preferences
KeyError: 'bpy_prop_collection[key]: key "slcad_transform" not found'

location: <unknown location>:-1

`

Skaifer, 2021-05-21:
Similar behavior here. 

Blender 2.93.0 Beta, Windows 10
Portable version installed with "Blender Launcher"
Tried both versions of addon

No menu items and tool options. Movement is working though. But when I switch addon off and on it shows this error:

```
 Python: Traceback (most recent call last):
  File "C:\Portable\Blender\Builds\daily\blender-2.93.0-beta+blender-v293-release.a294670bacb1-windows.amd64-release\2.93\scripts\startup\bl_ui\space_view3d.py", line 51, in draw
    self.draw_tool_settings(context)
  File "C:\Portable\Blender\Builds\daily\blender-2.93.0-beta+blender-v293-release.a294670bacb1-windows.amd64-release\2.93\scripts\startup\bl_ui\space_view3d.py", line 68, in draw_tool_settings
    tool = ToolSelectPanelHelper.draw_active_tool_header(
  File "C:\Portable\Blender\Builds\daily\blender-2.93.0-beta+blender-v293-release.a294670bacb1-windows.amd64-release\2.93\scripts\startup\bl_ui\space_toolsystem_common.py", line 782, in draw_active_tool_header
    draw_settings(context, layout, tool)
  File "C:\Users\7xrav\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\slcad_transform\slcad_transform.py", line 1042, in draw_settings
    prefs = context.preferences.addons[__package__].preferences
KeyError: 'bpy_prop_collection[key]: key "slcad_transform" not found'

location: <unknown location>:-1
```

s-leger, 2021-05-21:
CAD Transform will not provide support for blender beta version.
You may edit the file and change "blender addon" to "Blender addon" in order to fix this issue, or wait for next release.

Skaifer, 2021-05-21:
> CAD Transform will not provide support for blender beta version.
> You may edit the file and change "blender addon" to "Blender addon" in order to fix this issue, or wait for next release.

Oh, thanks. It's working now 😋

closed 2021-06-03 by s-leger