s-leger/blender_cad_transforms #62: error with w/ slcad_transform_0.93.2.beta2 and 2.93.1

theoryshaw, 2021-06-27:
w/ slcad_transform_0.93.2.beta2 and 2.93.1

when hitting 'g'. 

```
Python: Traceback (most recent call last):
  File "C:\Users\ryan\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\slcad_transform\slcad_transform.py", line 3189, in invoke
    self.init(context)
  File "C:\Users\ryan\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\slcad_transform\slcad_transform.py", line 3102, in init
    slcadsnap.start(context, prefs.snap_to_loose)
  File "C:\Users\ryan\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\slcad_transform\slcad_snap.py", line 223, in start
    self.update_gl_snap()
  File "C:\Users\ryan\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\slcad_transform\slcad_snap.py", line 697, in update_gl_snap
    self.gl_stack.set_snap_mode(self.snap_mode)
  File "C:\Users\ryan\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\slcad_transform\snap_context\__init__.py", line 235, in set_snap_mode
    self.update_all()
  File "C:\Users\ryan\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\slcad_transform\snap_context\__init__.py", line 223, in update_all
    self._offscreen.texture_color.clear(format='FLOAT', value=(0.0, 0.0, 0.0, 0.0))
AttributeError: 'GPUOffScreen' object has no attribute 'texture_color'

location: <unknown location>:-1

```

s-leger, 2021-06-27:
Version 0.93.2.beta is for blender 3.0 !
If you downgrade to 0.93.1 under blender 2.93, restart blender after setup.

theoryshaw, 2021-06-27:
cool!.. thx

closed 2021-06-27 by theoryshaw