s-leger/blender_cad_transforms #63: CAD transform 0.93.2 beta not working in edit mode on Blender 3.0

meikasahara, 2021-06-28:
After re-downloading the slcad_transform_0.93.2.beta2 from gumroad, the issue still persists in edit mode.

_Originally posted by @meikasahara in https://github.com/s-leger/blender_cad_transforms/issues/61#issuecomment-869420679_

meikasahara, 2021-07-04:
Using the recent daily 3.0 alpha blender builds, the problem persists with version slcad_transform_0.93.2.beta.3 for all snapping options in edit mode except for bounding box and 3d cursor.

s-leger, 2021-07-04:
CAD Transform will not snap on selected verts / faces in edit mode. So ensure target faces / vertex are not selected.

meikasahara, 2021-07-04:
The small red circle indicating the snapping does not appear on unselected vertices/faces either

s-leger, 2021-07-04:
Unable to repro here, mabe a display issue with too small circle radius for your screen ?
Any error in the console ?

https://user-images.githubusercontent.com/7442572/124380890-acbf1400-dcbf-11eb-8ec2-7fe3c2b3be24.mp4

meikasahara, 2021-07-04:
No errors in the console except for the occasional message 
draw_segs state False
when the cursor is near a vertex.

meikasahara, 2021-07-07:
Update: I have noticed the following behavior:
In edit mode, if
1) the mouse stays within the 3d viewport area, and
2) a transform key is pressed (for example G),
then snapping does not work, as previously shown.
However, if:
1) a transform key is pressed (e.g. G), and
2) the mouse, while the transform is active, hovers over a Spreadsheet editor and returns to the 3d viewport, then snapping works normally for that particular active transformation.
No warnings or errors are displayed in the console during this operation besides the occasional "draw_segs state False" (with logging set to "info").

closed 2021-09-20 by meikasahara