s-leger/blender_cad_transforms #65: CAD Transform takes over Object select

DarkVisions, 2021-07-24:
Hi guys,
Odd issue: when in Object mode, I can no longer use "normal" object tools to select Objects.  
Every time I choose the Select Tool in Object Mode and click on an object, CAD Transform tool is automatically selected for the existing object.  I try to change objects, and clicking the Select Tool works, but as soon as I click outside the selected object, on another object for example, it just changes tools back to CAD Transform. 
I tried disabling CAD Transform, and when I repeat the above steps, I get the error 
  "Tool 'slcad.transform' not found for space 'VIEW_3D'".  
So it looks like object selection is now tied to CAD Transform, and I can't get out.
So I'm stuck: I can't get it working, and I can't remove it.
Thoughts?

s-leger, 2021-07-24:
Hi,
Unable to repro here, but look like the tool system somewhat remains last selected tool and re-enable it ??
Any custom non standard addons enabled ?

Disable CAD Transform and restart blender.

DarkVisions, 2021-07-25:
Hi,
Yeah, it looks like the tool system keeps looking for the plugin, but this occurs even after I disable, remove and restart Blender: the tool select somehow tries to default back to the CAD Transform, even when it's completely removed, then can't find it at all and gives the error "Tool 'slcad.transform' not found for space 'VIEW_3D'", as noted above.  Given CAD Transform is completely removed at this point, it may be a Blender issue, rather than the plugin's fault?
With that in mind, I eventually fixed it by rolling back using the "Repair" option in Blender 2.92's install.  Interestingly, I didn't reinstall the CAD plugin, but it is present again, but now working as expected.
Odd error.  Thanks for taking the time to get back to me.

s-leger, 2021-07-25:
Hi, 
Given context (blender 2.92) and description of the issue, i guess the shortcuts where not properly unregistered, so blender complains about a missing tool when looking at shortcuts.

closed 2021-07-25 by s-leger