s-leger/blender_cad_transforms #67: View imperial dimensions in fractions.

theoryshaw, 2021-07-25:
Would be nice view imperial dimensions in fractions. 
AEC professional is US, rarely use the decimal feet display.

Perhaps there's a way to set a preference where it rounds the display to the nearest... 
1/64", 1/32", 1/16", 1/8", 1/4", 1/2", 1"

[image]

s-leger, 2021-07-25:
CAD Transform rely on blender's api for unit display, and is sensitive to scene units settings. 
Maybe "separate units" will help a bit.  
You should report to blender fundation (Jacques Luke).

theoryshaw, 2021-07-25:
cool will do. 
What's the proper channel for Blender feature requests?
- https://blender.community/c/rightclickselect/?sorting=hot or
- https://developer.blender.org/

s-leger, 2021-07-25:
I guess it is something for rightclickselect as it is not realy a bug.
You also may try to talk directly to Jacques Luke as he was in charge of last improvement of unit system.