s-leger/blender_cad_transforms #71: Can't specify/snap a base point on collection instance if the original collection is hidden.

MarcoHoo, 2021-12-16:
Can't specify/snap a base point on collection instance if the original collection is hidden.

https://user-images.githubusercontent.com/22971812/146302470-a8f976d2-af02-4941-9646-a02019c522ce.mp4

s-leger, 2021-12-16:
CAD Transform rely on blender's api scene ray cast for snap to mesh.
Would call collection instances not visible to ray cast when source collection is hidden a blender bug.
Could fix such issues with overcomplicated heavyweight setup tricks, but would be pretty inefficient and likely to slow down snap operations startup by seconds, so not an option to me.

MarcoHoo, 2021-12-18:
OK, I'm not going to use collection instance any more.

MarcoHoo, 2021-12-18:
Sir, your cad transform add-on is great. Do you have any interest in developing this feature [https://blender.community/c/rightclickselect/DqKy/](https://blender.community/c/rightclickselect/DqKy/ )
We can do more precision modeling with cad transform and definable orientation.
I don't konw your email so I ask here. Thank you. Sincerely. @s-leger