s-leger/blender_cad_transforms #84: All Transforms operations stopped working suddenly ....... with or without CAD Transform turned on..... what shall I do ?

JB00001, 2022-05-28:
Please help .... it is one of the best tool i have ever used in blender :)

MazerMart, 2022-06-09:
I took a screen shot of the error message.  
[image]

s-leger, 2022-06-09:
Thnx for your report, looks like a precision issue in "snappable item" index retrival code.
If you are unable to move anything once the error occurs, you must restart blender, there is no other way to actually make it work again.

MazerMart, 2022-06-09:
I have restarted several times and each time it happens again.  However, it
doesn't seem to happen on a simple cube.. Could it be something in the
Blender model.  This is the first time it has happened.