s-leger/blender_cad_transforms #89: Rotations on any other axis than z unreliable - dont snap to tagets

danielkrajnik, 2022-08-10:
I would like to review rotations on other axes. It seems to be a bug that:
Picking a pivot point
Picking a reference point
Choosing axis (very chaotic, model jumps a lot, no idea which axis it is without trying them all, and i mean all)
Picking a target point

The model ends up in a different, often completely unrelated angle. It would be extremely helpful to get this right.

s-leger, 2022-08-10:
Basically when you setup reference axis using a single point z axis (default rotation axis) will point up, y axis will be perpendicular to z axis and lie on xy plane, and x axis perpendicular to both y and z axis.

If you setup the rotation axis using an edge as reference, z axis will follow your edge, y axis will be perpendicular to z axis and lie on xy plane, and x axis perpendicular to both y and z axis.

So for some complex rotations you'll probably need 2 rotations.

You should take a look at Maker Tales tutorials about CAD Transform, could help a lot to figure out what's going on.
https://www.youtube.com/watch?v=dSjeqRy3KS0

danielkrajnik, 2022-08-10:
Thabk you for the prompt reply and for your work on this tool in general - it's been a huge help! 

I have watched all the videos from Maker Tales, however this problem with rotations hasn't been addressed. Objects may have their origins oriented in different ways and so X Y and Z may be different for different pairs of objects, the only way to know it is to go through all 9 combinations which involves many more steps than if we could just choose a second axis - I don't know if it's technically feasible?

danielkrajnik, 2022-08-10:
Please see a detailed explanation visualised below:

[image]
[image]
[image]
[image]
[image]
[image]

s-leger, 2022-08-10:
Use "E" in order to setup your Z axis using an edge snap at once, then snap to "from" and "to" should rotate your model the way you like.

danielkrajnik, 2022-08-10:
Thanks again, but how do you choose a "from" point? Once "E" edge is chosen it jumps immediately to "to" point. 

Images showing the issue:
[image]
[image]

s-leger, 2022-08-10:
In such situation, you'll have to add a temporary vertex on the monkey, so duplicate the eye and move it along local z axis.
Then you'll be able to setup your rotation axis using eyes points.
This will rotate along 1 axis, you may then adjust rotation along the other axis (using monkey eyes)

s-leger, 2022-08-10:
Alternative is to rotate the cube, so it does fith with the monkey, then rotate both back.

danielkrajnik, 2022-08-10:
Sorry, what I meant is that without being able to specify a "from" point that rotation couldn't be performed as intended. I'm not quite sure how creating another axis between the eyes would help? Rotating the target cube isn't always possible with more complex target objects.

Is there any other solution that could let you specify four points, two axes or an axis and two points that would be possible with this plugin. 

It's a really helpful tool and except for this issue with rotations it would be "pivotal" for CAD professionals

s-leger, 2022-08-10:
Not at this time, the main issue here is that you must specify the axis using the object you want to rotate (as from depends on that axis).
The quickest way to add a reference axis when missing (like in the monkey situation) is to actually extrude a verted to add an edge matching with desired "from" axis.

danielkrajnik, 2022-08-10:
I have tried that but unfortunately it still doesnt seem to work (monkey is rotated incorrectly). Please see screenshots below.

On the last image rotation is more than the picked target point.

[image]
[image]
[image]

s-leger, 2022-08-10:
https://user-images.githubusercontent.com/7442572/183929544-e65b97ad-dd29-4638-aa22-0cecc1334ce6.mp4

You may need 2nd rotation if your cube is not aligned with axis, but overall should work with temporary reference point.

danielkrajnik, 2022-08-10:
Thank you, that clarifies a lot. The problem is that it works with global axis, but causes the issue I described above when clicking z (x in my case) 3 times. First for the global then the second two to switch to the monkey's local axis. 

I believe that you could reproduce this if the cube was rotated at an arbitrary angle on all axes and tried to rotate the monkey then to one of its faces.

danielkrajnik, 2022-08-10:
Thanks a lot, that clarified that there was a problem with an object - I am not sure what causes it though. If you could take a look at the video below - one monkey isnt rotating like on the video you sent, while another one is. Both have the same orientation of the origins.

danielkrajnik, 2022-08-10:
As a follow up question are you aware of any other plugin that allows for a rotation with an arbitrary axis? I can see now that cad transform is meant to work only with object axes, but there are situation in professional CAD modelling where this is not enough.

danielkrajnik, 2022-08-10:
Sample file - if anyone knew how to rotate that with existing toolkit, that would solve the issue:
https://github.com/danielkrajnik/samples/blob/56fd623507e29ce510554621257f5686f677ab47/sample-rotation-problem.blend

danielkrajnik, 2022-09-01:
Follow up - still can't find a way to rotate on object's axis. Very much in demand.