s-leger/blender_cad_transforms #91: Is there a way to set the snap indicator stroke thickness?

gamecubate, 2022-09-28:
The add-on's preferences don't seem to include a property for this ; would this be something set elsewhere in Blender?

gamecubate, 2022-09-28:
The `Snap widgets line thickness` setting seems to have no effect. Would this setting have been it?

[image]

... on Blender 3.3.0 w. 0.93.2 beta 3 add-on version on macOS 10.15.7.

s-leger, 2022-09-28:
Yes, i guess there is something wrong in blender's python gpu api.
Will check on my side soon.

gamecubate, 2022-09-28:
Not sure this will help but my device (Macbook Pro) seems to be incompatible with Blender GPU rendering requirements (I am unable to set Preferences > System > Cycles Renders to Metal or any other GPU).