s-leger/blender_cad_transforms #97: CAD Transform in Blender 3.5,

klgc2, 2023-03-20:
When I click the box to enable CAD Transform in Blender 3.5, blender  lists a half dozen file paths then "Exception: Shader compile error." I can't find a way to copy the output, and the output disappears if I try to take a screen shot. 
I'm on a MacBook Pro (2.6 GHz Quad-Core intel Core i7) using MacOS 12.6.3. I don't have another computer to test this on, just the two blender versions (3.4.1 and 3.5).

klgc2, 2023-03-20:
I'm trying Blender 3.5 Beta.

s-leger, 2023-03-30:
You must enable "OpenGl" backend in preferences -> system instead of "Metal" one.
see https://projects.blender.org/blender/blender/issues/106304

zakkhoyt, 2023-04-01:
> You must enable "OpenGl"

That did work. Bummer to have to give up Metal but can make it work for now. Thx

jikometrix, 2023-05-04:
>> enable "OpenGl" backend in preferences -> system instead of "Metal"
This worked for me too.

dr-igor, 2023-05-21:
> You must enable "OpenGl" backend in preferences -> system instead of "Metal" one. see https://projects.blender.org/blender/blender/issues/106304

Based on the issue you linked to, this is due to this add-on using the old API - is there a plan to update the add-on to use the new API?

amby, 2023-06-16:
I've spent some time and updated shaders to make them work with Metal. Can't test OpenGL though, but should be fine as well. Haven't tested the whole functionality, but basics things like move, rotate, scale work fine.

Attached fixed files.

[fixes.zip](https://github.com/s-leger/blender_cad_transforms/files/11771673/fixes.zip)