s-leger/blender_cad_transforms #100: CAD Transform translation using plane-constraint causes Blender to freeze

jonnyns, 2023-04-16:
I do not know if this is a rather seldom used constraint, but this action causes blender to freeze:

Simple test from a freshly started Blender (same with 3.4.1 or 3.3.5) under Windows 10:

Select the cube.
- G to grab
- shift Z to choose XY-plane constraint
- LMB to select from-point
- move mouse to transform towards the back of the view (in the direction from the user)

If you move far enough, Blender freezes. Seems to be the same problem whichever constraint plane is selected (XY, XZ, YZ).

BTW. If you zoom out somewhat, then movement in other directions will also cause freezing.

s-leger, 2023-04-16:
Might be related to an AMD GPU causing shader recompilation related issue ?

jonnyns, 2023-04-16:
Not really sure what you are suggesting here... I am running off an Intel machine with a Nvidia 1080 Ti board. Using version slcad_transform_0.95.0. I have tested on several machines, all behave the very same way, freezing up Blender.
Did you verify the behaviour? Shold be rather quick to see...

s-leger, 2023-04-17:
Nope, i'm not running any w11 myself, last w thing was a w9, i'm running linux and macos.
Last reported slowdowns where related to a bug in blender's amd kernels, so it was the first thing i wanted to know.
That being said, at least under linux i'm able to repro that issue and will try to figure what's going wrong. 
Thank you for your report.