s-leger/blender_cad_transforms #103: CAD Transforms won't activate on Blender 4.0 alpha

meikasahara, 2023-05-27:
The latest Blender 4.0 alpha builds (from May 23 onwards) breka compatibility with the add-on, and as a result it can no longer be used or activated. Upon attempting to activate it, the following error appears in the console: 
`addon_utils.disable: slcad_transform not disabled
Traceback (most recent call last):
  File "/home/user/Blender/blender-4.0.0-alpha+main.8ed65fe6de9a-linux.x86_64-release/4.0/scripts/modules/addon_utils.py", line 333, in enable
    mod = __import__(module_name)
  File "/home/user/.config/blender/4.0/scripts/addons/slcad_transform/__init__.py", line 63, in <module>
    from .slcad_transform import register, unregister
  File "/home/user/.config/blender/4.0/scripts/addons/slcad_transform/slcad_transform.py", line 45, in <module>
    from .utils.gpu_draw import (
  File "/home/user/.config/blender/4.0/scripts/addons/slcad_transform/utils/gpu_draw.py", line 401, in <module>
    class GPU_2d_uniform(GPU_Draw):
  File "/home/user/.config/blender/4.0/scripts/addons/slcad_transform/utils/gpu_draw.py", line 402, in GPU_2d_uniform
    _shader = get_shader('2D_UNIFORM_COLOR')
  File "/home/user/.config/blender/4.0/scripts/addons/slcad_transform/utils/gpu_draw.py", line 85, in get_shader
    return gpu.shader.from_builtin(builtin_typ)
ValueError: expected a string in ('FLAT_COLOR', 'IMAGE', 'IMAGE_COLOR', 'SMOOTH_COLOR', 'UNIFORM_COLOR', 'POLYLINE_FLAT_COLOR', 'POLYLINE_SMOOTH_COLOR', 'POLYLINE_UNIFORM_COLOR'), got '2D_UNIFORM_COLOR'

`

s-leger, 2023-05-27:
Thank you for reporting.

meikasahara, 2023-05-29:
It seems for Blender 4.0 all strings 3D_UNIFORM_COLOR should be replaced with UNIFORM_COLOR according to the example in the docs - https://docs.blender.org/api/4.0/gpu.html#d-lines-with-single-color

There are probably others incompatible changes in 4.0 as well.

SecondGM, 2023-07-12:
Can't make it work. It still crashes a lot while using, even with those lines corrected.