s-leger/blender_cad_transforms #104: Error while using the plugin

SecondGM, 2023-06-26:
[image]

sz6vqg, 2023-06-26:
Hi, I got the same error message when trying to install "slcad_transform_0.95.0" in Blender 3.5.0 (on Mac OS Ventura 13.4.1, running on M1 hardware). The same version "slcad_transform_0.95.0" installes correctly in Blender 3.5.0 on Windows 10.

[image]

gurucc, 2023-07-05:
I get the same - M2 Max MacBook Pro on Blender 3.5.1 as sz6vqg - line 333, exact same error. 

Traceback (most recent call last):
File "/Applications/Blender.app/Contents/Resources/3.5/scripts/modules/addon_utils.py", line 333, in enable
mod = _ import_(module _name)
File "/Users/ben/Library/Application Support/Blender/3.5/scripts/addons/sicad_transform/_ init_py", line 63, in <module> from .sicad transform import register, unregister
File "/Users/ben/Library/Application Support/Blender/3.5/scripts/addons/sicad transform/sicad transform.py", line 45, in «module> from utils.gpu_draw import (
File "/Users/ben/Library/Application Support/Blender/3.5/scripts/addons/sicad_transform/utils/gpu_draw.py", line 289, in <module> class GPU_3d_uniform(GPU_Draw):
File "/Users/ben/Library/Application Support/Blender/3.5/scripts/addons/sicad transform/utils/gpu draw.py", line 291, in GPU 3d uniform
_shader = get shader)
File "/Users/ben/Library/Application Support/Blender/3.5/scripts/addons/sicad transform/utils/gpu draw.py", line 87, in get shader return gputypes.GPUShader(vertex 3d, fragment flat color)
Exception: Shader Compile Error, see console for more details

s-leger, 2023-07-05:
https://github.com/s-leger/blender_cad_transforms/issues/97#issuecomment-1490556657

closed 2023-07-05 by s-leger