s-leger/blender_cad_transforms #105: CAD Transforms completely freezing Blender, and Blender never recovers

JoshuaHarris1, 2023-06-27:
Hi,

I have been struggling with this issue for a week or two now and have been unable to figure it out. I am relatively new to blender, so apologies if it ends up being something obvious...

I have a model I am making for a work project, and upon invoking CAD Transforms to move an object with snap to edge centre, the entire application freezes. I do not seem to get an error log anywhere, have tried running it in debug mode from console on each device, but nothing. The application just stops responding and never recovers again. Once in a blue moon it CAN recover, but 95% of the time it doesn't. I have tried this on multiple devices, my quite powerful high end gaming PC at home.. a relatively powerful gaming laptop and my work laptop, it seems to be happening on each one. In terms of file size I feel this shouldn't be an issue either... it is less than 1mb currently when it is having these issues. 

I very much appreciate any advice anyone can give me, and can provide additional info/screenshots/files if need be. This issue has been causing problems for a little while now like I mentioned and I just can't figure it out and am on my last legs with it lol

SecondGM, 2023-07-17:
Same here and for a while. At this point I just believe the pluggin is dead?

FeuerTiger86, 2023-07-17:
I have the same problem in Blender 3.3.8 On some objects, moving or rotating completely freezes Blender with v0.95.0 of the addon. However, I went back to v0.93.2, and for now if appears to be working.

s-leger, 2023-07-17:
CAD Transform is in complete rewrite from scratch as over the time it becomes difficult to manage / extend, so until then would avoid to loose time in changes on code base.

deedend, 2023-07-17:
Great to hear that @s-leger!! I find your add-on incredibly useful, is it possible to know where can I follow the development or at least get a notification when it will be released? Thanks for your great work, it's a life saver for our poor CAD designer that try to use blender!

gamecubate, 2023-08-02:
Same here. I use it everyday for a cabin construction project of ours and, frankly, I would hesitate to use Blender without it.  Charge for this and I will pay. That said, it doesn't seem Blender 3.6.0 LTS friendly. When checking the activation box in the Add-Ons section of Settings, the process fails with the following :
```
Traceback (most recent call last):
  File "/Users/alexandre.rousseau/local/apps/Blender-3.6.1.app/Contents/Resources/3.6/scripts/modules/addon_utils.py", line 333, in enable
    mod = __import__(module_name)
  File "/Users/alexandre.rousseau/Library/Application Support/Blender/3.6/scripts/addons/slcad_transform/__init__.py", line 63, in <module>
    from .slcad_transform import register, unregister
  File "/Users/alexandre.rousseau/Library/Application Support/Blender/3.6/scripts/addons/slcad_transform/slcad_transform.py", line 45, in <module>
    from .utils.gpu_draw import (
  File "/Users/alexandre.rousseau/Library/Application Support/Blender/3.6/scripts/addons/slcad_transform/utils/gpu_draw.py", line 289, in <module>
    class GPU_3d_uniform(GPU_Draw):
  File "/Users/alexandre.rousseau/Library/Application Support/Blender/3.6/scripts/addons/slcad_transform/utils/gpu_draw.py", line 291, in GPU_3d_uniform
    _shader = get_shader()
  File "/Users/alexandre.rousseau/Library/Application Support/Blender/3.6/scripts/addons/slcad_transform/utils/gpu_draw.py", line 87, in get_shader
    return gpu.types.GPUShader(vertex_3d, fragment_flat_color)
Exception: Shader Compile Error, see console for more details
```.

SwissmountainVA, 2023-10-28:
Same problem with Blender hanging from time to time in Windows Blender 3.6.1 

I like the functionality of this plugin and would pay for it. Will this be updated any time soon?

Thank you for this addon.

s-leger, 2023-10-28:
A total refactory of CAD Transform is in closed beta state and will be released soon.

SwissmountainVA, 2023-10-29:
@s-leger thank you for the heads-up!