s-leger/blender_cad_transforms #107: CAD Transform Add On could not be activated, Blender 3.5.1

PaulMeier007, 2023-08-10:
System Information
Operating system: MacBook Air, 2020, macOS Monterey 12.6.2
Graphics card: Intel Iris Plus Graphics 1536 MB

Blender Version
Broken:Version 3.5.1 (3.5.1 2023-04-25)
Worked: none

Short description of error
Hello there,

I tried to install the Add On CAD Transform in Blender 3.5.1 The installation is no problem but when I try to activate it, I get an error message. Did somebody else experienced the same or knows how to solve it? There is an Issue with CAD Sketcher, too ...

Thank you upfront

Total console output is:
Traceback (most recent call last):
  File "/Applications/Blender.app/Contents/Resources/3.5/scripts/modules/addon_utils.py", line 333, in enable
    mod = __import__(module_name)
  File "/Users/Julius/Library/Application Support/Blender/3.5/scripts/addons/slcad_transform/__init__.py", line 63, in <module>
    from .slcad_transform import register, unregister
  File "/Users/Julius/Library/Application Support/Blender/3.5/scripts/addons/slcad_transform/slcad_transform.py", line 45, in <module>
    from .utils.gpu_draw import (
  File "/Users/Julius/Library/Application Support/Blender/3.5/scripts/addons/slcad_transform/utils/gpu_draw.py", line 289, in <module>
    class GPU_3d_uniform(GPU_Draw):
  File "/Users/Julius/Library/Application Support/Blender/3.5/scripts/addons/slcad_transform/utils/gpu_draw.py", line 291, in GPU_3d_uniform
    _shader = get_shader()
  File "/Users/Julius/Library/Application Support/Blender/3.5/scripts/addons/slcad_transform/utils/gpu_draw.py", line 87, in get_shader
    return gpu.types.GPUShader(vertex_3d, fragment_flat_color)
Exception: Shader Compile Error, see console for more details

Inok007, 2023-08-17:
The same problem. 
iMac Silicon 24''.

s-leger, 2023-08-17:
https://github.com/s-leger/blender_cad_transforms/issues/97#issuecomment-1490556657

closed 2023-08-17 by s-leger