s-leger/blender_cad_transforms #110: CAD transform causes blender 4.0.1 crash

linuxuser63, 2023-11-23:
Just downloaded blender 4.0.1 and everytime I open it it crashes. Wrote to blender and they said that based on the info provided it appeared that "slcad_transform" was the issue. I turned CAD Transform off and reloaded BL 4.0.1 and it worked and did not crash.  Using CAD Transform 0.95.0

Crash information from command line:
File "/home/lgoldman/.config/blender/4.0/scripts/addons/slcad_transform/init.py", line 63, in
from .slcad_transform import register, unregister
File "/home/lgoldman/.config/blender/4.0/scripts/addons/slcad_transform/slcad_transform.py", line 45, in
from .utils.gpu_draw import (
File "/home/lgoldman/.config/blender/4.0/scripts/addons/slcad_transform/utils/gpu_draw.py", line 401, in
class GPU_2d_uniform(GPU_Draw):
File "/home/lgoldman/.config/blender/4.0/scripts/addons/slcad_transform/utils/gpu_draw.py", line 402, in GPU_2d_uniform
_shader = get_shader('2D_UNIFORM_COLOR')
File "/home/lgoldman/.config/blender/4.0/scripts/addons/slcad_transform/utils/gpu_draw.py", line 85, in get_shader
return gpu.shader.from_builtin(builtin_typ)
ValueError: expected a string in ('FLAT_COLOR', 'IMAGE', 'IMAGE_COLOR', 'SMOOTH_COLOR', 'UNIFORM_COLOR', 'POLYLINE_FLAT_COLOR', 'POLYLINE_SMOOTH_COLOR', 'POLYLINE_UNIFORM_COLOR'), got '2D_UNIFORM_COLOR'
blenderkit updater: Running background check for update
blenderkit addon: Starting background checking thread
blenderkit addon: Checking for update now in background
blenderkit addon: Checking for update function
blenderkit INFO: Global dir permissions are OK. [14:09:15.585, utils.py:1296]
blenderkit addon: Read in JSON settings from file
Writing: /tmp/blender.crash.txt
Segmentation fault (core dumped)

System info:
Operating System: Mageia 9
KDE Plasma Version: 5.27.5
KDE Frameworks Version: 5.105.0
Qt Version: 5.15.7
Kernel Version: 6.4.16-desktop-3.mga9 (64-bit)
Graphics Platform: X11
Processors: 8 × Intel® Core™ i7 CPU 950 @ 3.07GHz
Memory: 23.5 GiB of RAM
Graphics Processor: NVIDIA GeForce GTX 1050 Ti/PCIe/SSE2
Manufacturer: Gigabyte Technology Co., Ltd.
Product Name: X58A-UD5

s-leger, 2023-11-23:
CAD Transform 0.9x is not compatible with blender 4.x, there are many breaking changes in the blender's API.
You must wait for next release.

Ossitapa, 2023-11-27:
When it is expected to work again? pretty bummed that i bought it just now, and its not working. Should have even mentioned it, before hand..

s-leger, 2023-11-27:
Soon, version 2.0 is in closed beta state now.

In facts it is "half - mentioned" in product description.
`CAD like transform operations for blender 2.81+ / 2.91+ / 3.0+`

With breaking changes like Metal API and known weakness / issues in 0.9.x series it is obvious that simple fixes are not suited anymore to keep it working as expected.
So took the decision to completely rewrite from scratch, and upcoming release will be a true game changer.

helour, 2023-11-28:
> Just downloaded blender 4.0.1 and everytime I open it it crashes. Wrote to blender and they said that based on the info provided it appeared that "slcad_transform" was the issue. I turned CAD Transform off and reloaded BL 4.0.1 and it worked and did not crash. Using CAD Transform 0.95.0
 
> System info: Operating System: Mageia 9 KDE Plasma Version: 5.27.5 KDE Frameworks Version: 5.105.0 Qt Version: 5.15.7 Kernel Version: 6.4.16-desktop-3.mga9 (64-bit) Graphics Platform: X11 [...]

Try to change line 402 in the file /home/lgoldman/.config/blender/4.0/scripts/addons/slcad_transform/utils/gpu_draw.py to:
`_shader = get_shader('UNIFORM_COLOR')`

linuxuser63, 2023-11-28:
Thank you helour.  I tried your suggestion, but CAD transform would not open in blender 4.0.1
I appreciate the effort

fcrbe, 2023-12-12:
@s-leger Do you have an indication on when you would be releasing v2.0?

s-leger, 2023-12-12:
End of this week probably.

BarryDGleeson, 2023-12-22:
Can't wait

dallinlemon, 2023-12-23:
> End of this week probably.

@s-leger Do we have any updates on this? Where should we be watching for the latest updates/status? 
I am currently really excited to work through the PDT course, but I need to wait until I have the addon working.

Brinax, 2023-12-26:
> End of this week probably.

Can't wait

s-leger, 2024-01-06:
Released CAD Transform 2.0 for blender 3.6 / 4.x
https://blender-archipack.gumroad.com/l/emiwtw