s-leger/blender_cad_transforms #113: Snapping doesn't work in 2D (without faces) in Version 0.95.0

Chantico360, 2023-12-27:
Hello,

first of all.. thanks for the great addon! I'm learning blender using the Tutorials from Maker Tales and found a problem with the Addon in Version 0.95.0. The older version 0.93.2.beta.3 works.

If you create an object without a face (simple square in 2D) you cannot select or snap to anything (vertex, edge) other than the origin. It works in the older version.

Example:
[image]

I cannot snap-select any vertex or edge in the example. If I create faces, it works.

Regards,
Tobias

s-leger, 2024-01-06:
Broken by blender api changes.
Made a complete refactory from scratch to handle this properly - CAD Transform 2.0 for blender 3.3+ / 4+.
it is available on gumroad's store 
[CAD Transform 2.0](https://blender-archipack.gumroad.com/l/emiwtw)

Chantico360, 2024-02-03:
Thanks for your answer. I already bought CAD Transform 2.0 and I'm using it with Blender LTS 3.6.8.0. It works better in 2.0, but not as expected (am I using it wrong?).

Same shape like in the initial comment. It works on the outer edges, but not in the middle. If I create faces, the snapping works on the middle vertext. 

Form with all vertex are correct linked:
[image]

Outer vertex snapping works:
[image]

Middle vertex won't snap:
[image]

Middle vertex snap with faces:

[image]

s-leger, 2024-02-03:
In 2.0, you must enable snap to "Isolated elements" (rightmost snap option) in order to snap to edges / vertex not linked to faces.
It is disabled by default as it may be a bit slow with huge mesh objects.

Chantico360, 2024-02-03:
Thanks, that does the trick. Just one flaw: if I enable it during a operation (e.g. "G-"Key) it won't work until I restart the operation. That denies changing the snapping while operation (that was cool feature).

s-leger, 2024-02-04:
You may alter other snap options while running, but this one needs to be set first as it does affect mesh object initialization. You may alter snap options in preferences so it is always enabled.

Chantico360, 2024-02-04:
Ah okay. Thanks for the help and for the great addon!

closed 2024-02-04 by Chantico360